garmin/LIDARLite_Arduino_Library #2: Too many blocking code:(

Serveurperso, 2016-11-30:
Hello,

Very good spec sensor but the lib is useless for serious application.
There is too many blocking code inside it :

 while(busyFlag != 0) // Loop until device is not busy
{
... update busyFlag ...
}

 
delay(1); // 1 ms delay for robustness with successive reads and writes

You can't do this 1ms is a huge waste on a microcontroller.

If this sensor can't be used with serious non-blocking code, it's a crap.

Pascal

nseverson, 2016-11-30:
Hi Pascal,

Thanks for your interest in LIDAR-Lite! The purpose of this library is to help people get started using the sensor with an easy to understand interface. It is written for the Arduino platform because many people are already familiar with it, and it is intentionally as simple as possible.

User AlexisTM has written a [robust LIDAR-Lite library](https://github.com/AlexisTM/LIDAREnhanced) for Arduino that uses a non-blocking interface. This may be useful for your project.

Regards,

Neil Severson

closed 2016-11-30 by nseverson

Serveurperso, 2016-11-30:
Amazing !!! Thanks a lot for the AlexisTM information !!! (I not found with my google request)

AlexisTM, 2016-12-02:
Indeed, I am actually the last library from the search "lidarlite" :(

Serveurperso, 2016-12-02:
AlexisTM, a minimalist example is required inside your lib. Because no one can buy 5 or more Lidars to test it:) I open a doc issue when I make my first one lidar sample:)

AlexisTM, 2016-12-02:
Here it is : https://github.com/AlexisTM/LIDAREnhanced/blob/master/example/OneLaser/OneLaser.ino

I never tried on other chips... and I include Arduino.h to make continuous integration easier.

Serveurperso, 2016-12-02:
Very easy to integrate this lib inside my projet:) thanks a lot.
My project is a public web controlled robot online here : http://www.serveurperso.com/?page=robot

AlexisTM, 2016-12-03:
I am happy it worked so well :D

Anf that's a funny project (I just checked it)

Serveurperso, 2016-12-03:
// Trigger pin, can be unplugged
#define Z1_LASER_TRIG 11
// Enable pin, IMPORTANT
#define Z1_LASER_EN 12
// Mode pin, can be unplugged
#define Z1_LASER_PIN 13

Why you assign useless pin ? How I can get sure they are never used (if use near all I/O on my PIC32MX) ? Your lib is energy efficient for my battery (you use the "Enable" pin this is a very good idea) ?

Serveurperso, 2016-12-03:
There is a lot of documentation missing.... I must open a case on your github
What is the 'A'
What is the DISTANCE global ?
What is the 2 magic number ?

AlexisTM, 2016-12-03:
Hi, indeed, there are missing documentation as I am currently the only one using it, I have no feedback yet. :octocat: 

* You can "shutdown" the laser by calling `Controller::resetLidar(0);`. Once you'll call `Controller.spinOnce` again, it will start the procedure to wakeup the laser.
* Otherwise, If you do not call "Controller.spinOnce()", the laser will NOT take any acquisition.
* I do not know (not tested) how much energy it consumes.
* The "2" is the configuration, *forked* from the original library. 
* The 'A' is the name of the laser, allowing to know which laser called the callback (`if(self->name == 'A')`)
* The DISTANCE is a "to implement" enum, that would allow to take velocity measurements by replacing DISTANCE by VELOCITY or DISTANCE_AND_VELOCITY.

NOTE : The Trigger pin & Mode pin are not yet used. It could be useful for someone who wants to take measurements with PWM. Plus, I tried taking measurements with the "continuous" mode of the lasers (with trigger & mode pin), but it was strangely slower. Therefore, the mode pin & trigger pin are never used.

Serveurperso, 2016-12-03:
At this time I get a spinOnce freeze with wire on PIC32 last chipkit Core.
I temporarily run the lidar on 3.3v (not recommended) because there is no 5V on my robot at this time.

Debug enabled :

setup()
loop()
Laser 0 SHUTING_DOWN
Laser 0 NEED_RESET
Laser 0 RESET_PENDING
Laser 0 RESET_PENDING
Laser 0 RESET_PENDING
Laser 0 RESET_PENDING
Laser 0 RESET_PENDING
Laser 0 RESET_PENDING
Laser 0 RESET_PENDING
Laser 0 RESET_PENDING
Laser 0 RESET_PENDING
Laser 0 RESET_PENDING
Laser 0 RESET_PENDING
Laser 0 ACQUISITION_IN_PROGRESS

Serveurperso, 2016-12-03:
Same problem with a new 5V 1A Tracopower DC-DC. The stock lidar lib (this one) work.
Freeze :
SDL/SDA pins low
Enable pin high (3.3)

Serveurperso, 2016-12-04:
Now it work !!!

```
      //Wire.requestFrom(Device, uint8_t(2), uint8_t(1));
      Wire.requestFrom(Device, uint8_t(2));
      data[0] = Wire.read();
      data[1] = Wire.read();
```

I make a mistake I swapped uint8_t(2) and uint8_t(1) for readWord() / readByte() when I solved a requestFrom compatibility problem lol

Serveurperso, 2016-12-04:
It work for a time and freeze:(
Wire is unreliable
I must rewrite this directly in PIC32 DTWI to bypass the crap wire wrapper from chipKIT:'(

AlexisTM, 2016-12-04:
I agree for Wire ... Especially for the Arduino Due.

There are probably already Wire replacement libraries for your chip! (Note:
did you tested with 100kHz I2C instead of 400kHz?)

If you need to modify I2C functions, almost all are in the I2CFunctions
file and there is one occurrence in Controller::begin, which start Wire.

https://github.com/AlexisTM/LIDAREnhanced/blob/master/I2CFunctions.h

Serveurperso, 2016-12-04:
Yes I run I2C @ 400KHz I already commented out the low level configuration line inside your code. Because I have a define in my Digilent DTWI header.

There is a I2C vs another library conflict and a timer/interrupt problem.

Serveurperso, 2016-12-04:
I opened an issue on chipKIT Core API
https://github.com/chipKIT32/chipKIT-core/issues/312