garmin/LIDARLite_Arduino_Library #3: Arduino IDE  ver. 18 and LidarLite V3 Library (version 1.02)

ksmith9109, 2017-01-04:
I downloaded the Garmin Arduino Library  **LIDARLite_v3_Arduino_Library-1.0.2.zip.**

I added it via  Add ZIP Library command.

Then I get the errors listed below:

**Invalid version  found:  v1.0.2**

Also
Library Manager states that installed library is verion 1.0.0

Also the library manager "More Info"  link does not allow any Version Install xxxxxx  option.
The "More Info "  link just takes you to the garmin store and offers to sell the Lidar Lite V 3 hardware.
    I plan on purchasing the Lidar Lite V3 but want to make sure the library updates are correct.

Thanks for your help in advance.
K Smith

[image]

Lauszus, 2017-01-05:
I believe the problem is that the version should be ```1.0.2``` and NOT ```v1.0.2``` here: https://github.com/garmin/LIDARLite_v3_Arduino_Library/commit/fd33e612589e6f52208f9443a41942c32ab71da8#diff-ea32f5b85735296fd64b577e91a19607R2. The same goes for the tags.

Lauszus, 2017-01-05:
But since there is actually no code change since ```1.0.0``` you should be fine: https://github.com/garmin/LIDARLite_v3_Arduino_Library/compare/1.0.0...v1.0.2?w=1.

nseverson, 2017-01-25:
Thanks. Updated version to 1.0.3 to play nice with the Arduino Library Manager.

closed 2017-01-25 by nseverson