garmin/LIDARLite_Arduino_Library #15: Lidar Lite v3 not working

kpihus, 2019-10-23:
Hi, 

I received a brand new Lidar-Lite v3 (2019/05) yesterday.
Hooked up to Arduino Leonardo by specification shown in thos document
http://static.garmin.com/pumac/LIDAR_Lite_v3_Operation_Manual_and_Technical_Specifications.pdf
Green to Arduino SCL pin, blue to SDA pin. The whole setup gets power from external 5V BEC with 680 uF capacitor in parallel.

I tried with example code in this library, but all that I get is nack in the serial terminal.

Is there anything special that I have to do besides that, or any guidelines on how to debug this issue further?

--
BR,
Kpihus

dwinters42, 2019-10-23:
Just to make sure it is nothing trivial: Have you connected GND of the BEC with GND on the LidarLite _and_ the Arduino?

kpihus, 2019-10-23:
Yes, Arduino gets also power from that same BEC over 5V and GND pins

kpihus, 2019-10-23:
I did one more test and hooked it up to Pixhawk controller with PX4 software and PWM mode. Still nothing.