garmin/LIDARLite_Arduino_Library #21: Possible to read Control Registers in PWM mode?

Eileenmc, 2020-06-30:
We're using the Lidar Lite v3 HP configured in PWM to an Arduino Uno and would like to be able to read the control registers, for example, Health Status in the serial monitor. Is this possible? Looking through the Operations Manual and Tech specs isn't clear if the registers can only be used in I2C mode. Thank you in advance for your help with this!

Eileenmc, 2020-07-01:
Gave some thought to this last night after I posted the above question. The issue is that we want to use the lidar in PWM mode but also want to be able to read the registers while it's running.   Is the solution using two arduino's  -  one to communicate with the Lidar in PWM as the master and the other communicating with it in I2C as slave?

jmseitz, 2020-07-01:
You can utilize TRIGGER / MONITOR connections and I2C connections to the v3HP, simultaneously. The slave peripheral inside the v3HP continues to operate during PWM mode, allowing full access to the device's register set. You could use a separate Arduino to do this if you prefer, but both interfaces can be managed by a single Arduino.