garmin/LIDARLite_Arduino_Library #24: LIDAR Lite V3 HP with ESP32

Speedy2k, 2020-11-19:
Currently it seems that the ESP32 wire library doesn support the 5 params requestFrom() command. So as it is now, your library isn't compatible with ESP32. I have modified your read() function to be compatible on the ESP32, i am unshure it is the best way to do it, but it is working on my setup.!

`void LIDARLite_v3HP::read(uint8_t regAddr,  uint8_t * dataBytes,
                          uint8_t numBytes, uint8_t lidarliteAddress)
{
    // This single function performs the following actions -
    //     1) I2C START
    //     2) I2C write to set the address
    //     3) I2C REPEATED START
    //     4) I2C read to fetch the required data
    //     5) I2C STOP
    Wire.beginTransmission(lidarliteAddress);
    Wire.write(regAddr);
    if(!Wire.endTransmission()){
		Wire.requestFrom
		(
			lidarliteAddress, // Slave address
			numBytes,         // number of consecutive bytes to read
			//regAddr,          // address of first register to read
			//1,                // number of bytes in regAddr
			true              // true = set STOP condition following I2C read
		);

		uint8_t  numHere = Wire.available();
		uint8_t  i       = 0;
		while (i < numHere)
		{
			dataBytes[i] = Wire.read();
			i++;
		}
		delayMicroseconds(100); // 100 us delay for robustness with successive reads and writes
    }else{
		Serial.println("> nack");
    }
} /* LIDARLite_v3HP::read */`

BradWiseman, 2020-11-19:
Speedy2k, thanks for providing the details on what you did to modify our Arduino library code for compatibility to the ESP32 wire library.    Garmin provided the LIDAR-Lite Arduino Library as an example code base that can be ported to programming environments other than Arduino UNO.  We appreciate it when people share examples of what it took to port our Arduino UNO example code to support a different programming environment!  I hope that your example will be useful to others who wish to port Garmin's Arduino UNO code for use on another platform.

Speedy2k, 2020-11-19:
Justbto let you t’know that the modification is for arduino and should also be compatible with the Uno!

gtestault, 2020-11-26:
According to the arduino reference there is no 5 parameter version of the requestFrom version in the Wire library: [https://www.arduino.cc/en/Reference/WireRequestFrom](https://www.arduino.cc/en/Reference/WireRequestFrom)

Could you please fix the library upstream because the same issue is not making the library work for the Arduino BLE 3.3v too

BradWiseman, 2020-12-03:
gtesault, you'll notice new code added to the LIDARLite_v3HP.cpp source code that introduces a new compiler switch, LEGACY_I2C.  If this variable is defined, the compiler will choose the standard 3 parameter version of the requestFrom() function.  Otherwise, it defaults to the 5 parameter version.  Here are the notes that were submitted with this improvement to the LIDAR-Lite v3HP source code:

Arduino Due needs to the overloaded version of requestFrom() to
stabilize and to perform repeated STARTs. The LEGACY_I2C added
here gives users a method to continue to use this library if
their selected processor does not implement the overloaded function.

The new code resides here: https://github.com/garmin/LIDARLite_Arduino_Library/commit/9c68d9089d0e654a5c2aafcb8e938a2a94498d53

maulepilot, 2020-12-06:
BradWiseman, I haven't been able to get this library to compile for the LIDAR-Lite V3 HP v3HP/v3HP_I2C sketch. I'm using an Arduino MKR WIFI 1010 board. The verbose compiler messages indicate that the problem is a call to the requestFrom function in the TwoWire class with 5 parameters instead of 3. Per your instructions, I created a platform.local.txt file with the single line

build.extra_flags=-DLEGACY_I2C

in the same folder as platform.txt on my Mac but the code will still not compile. Here are the error messages from the compiler. Any advice?

Arduino: 1.8.13 (Mac OS X), Board: "Arduino MKR WiFi 1010"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/ameranda/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/ameranda/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/ameranda/Documents/Arduino/libraries -fqbn=arduino:samd:mkrwifi1010 -vid-pid=2341_8054 -ide-version=10813 -build-path /var/folders/wq/32dsft1s2zngpbfvg_lqvjn40000gn/T/arduino_build_872686 -warnings=none -build-cache /var/folders/wq/32dsft1s2zngpbfvg_lqvjn40000gn/T/arduino_cache_540087 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=/Users/ameranda/Library/Arduino15/packages/arduino/tools/bossac/1.7.0-arduino3 -prefs=runtime.tools.bossac-1.7.0-arduino3.path=/Users/ameranda/Library/Arduino15/packages/arduino/tools/bossac/1.7.0-arduino3 -prefs=runtime.tools.CMSIS-Atmel.path=/Users/ameranda/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=/Users/ameranda/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0 -prefs=runtime.tools.CMSIS.path=/Users/ameranda/Library/Arduino15/packages/arduino/tools/CMSIS/4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=/Users/ameranda/Library/Arduino15/packages/arduino/tools/CMSIS/4.5.0 -prefs=runtime.tools.arm-none-eabi-gcc.path=/Users/ameranda/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=/Users/ameranda/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.openocd.path=/Users/ameranda/Library/Arduino15/packages/arduino/tools/openocd/0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=/Users/ameranda/Library/Arduino15/packages/arduino/tools/openocd/0.10.0-arduino7 -prefs=runtime.tools.arduinoOTA.path=/Users/ameranda/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=/Users/ameranda/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.2.1 -verbose /Users/ameranda/Documents/Arduino/libraries/LIDARLite_Arduino_Library-3.0.5/examples/v3HP/v3HP_I2C/v3HP_I2C.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/ameranda/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/ameranda/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/ameranda/Documents/Arduino/libraries -fqbn=arduino:samd:mkrwifi1010 -vid-pid=2341_8054 -ide-version=10813 -build-path /var/folders/wq/32dsft1s2zngpbfvg_lqvjn40000gn/T/arduino_build_872686 -warnings=none -build-cache /var/folders/wq/32dsft1s2zngpbfvg_lqvjn40000gn/T/arduino_cache_540087 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=/Users/ameranda/Library/Arduino15/packages/arduino/tools/bossac/1.7.0-arduino3 -prefs=runtime.tools.bossac-1.7.0-arduino3.path=/Users/ameranda/Library/Arduino15/packages/arduino/tools/bossac/1.7.0-arduino3 -prefs=runtime.tools.CMSIS-Atmel.path=/Users/ameranda/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=/Users/ameranda/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0 -prefs=runtime.tools.CMSIS.path=/Users/ameranda/Library/Arduino15/packages/arduino/tools/CMSIS/4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=/Users/ameranda/Library/Arduino15/packages/arduino/tools/CMSIS/4.5.0 -prefs=runtime.tools.arm-none-eabi-gcc.path=/Users/ameranda/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=/Users/ameranda/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.openocd.path=/Users/ameranda/Library/Arduino15/packages/arduino/tools/openocd/0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=/Users/ameranda/Library/Arduino15/packages/arduino/tools/openocd/0.10.0-arduino7 -prefs=runtime.tools.arduinoOTA.path=/Users/ameranda/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=/Users/ameranda/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.2.1 -verbose /Users/ameranda/Documents/Arduino/libraries/LIDARLite_Arduino_Library-3.0.5/examples/v3HP/v3HP_I2C/v3HP_I2C.ino
Using board 'mkrwifi1010' from platform in folder: /Users/ameranda/Library/Arduino15/packages/arduino/hardware/samd/1.8.9
Using core 'arduino' from platform in folder: /Users/ameranda/Library/Arduino15/packages/arduino/hardware/samd/1.8.9
[... cut ...]
             ^~~~~~~~~~~
/Users/ameranda/Library/Arduino15/packages/arduino/hardware/samd/1.8.9/libraries/Wire/Wire.h:46:13: note:   candidate expects 2 arguments, 5 provided
Using library Wire at version 1.0 in folder: /Users/ameranda/Library/Arduino15/packages/arduino/hardware/samd/1.8.9/libraries/Wire 
Using library LIDARLite_Arduino_Library-3.0.5 at version 3.0.5 in folder: /Users/ameranda/Documents/Arduino/libraries/LIDARLite_Arduino_Library-3.0.5 
exit status 1
Error compiling for board Arduino MKR WiFi 1010.

BradWiseman, 2020-12-08:
@maulepilot: Please try replacing this line:

   build.extra_flags=-DLEGACY_I2C

with this line:

   compiler.cpp.extra_flags=-DLEGACY_I2C

maulepilot, 2020-12-08:
The V3HP_I2C and V3HP_MONITOR example sketches will still not compile after making this change. I attached the verbose compiler output for the V3HP_I2C sketch. To make sure that I am following your instructions correctly, here's what I did. I created the file  /Users/ameranda/Library/Arduino15/packages/arduino/hardware/samd/1.8.9/platform.local.txt which contains the single line:

compiler.cpp.extra_flags=-DLEGACY_I2C

I know that the Wire.h library works correctly with the Arduino MKR WIFI 1010 board because I have an AMG8833 thermal camera example sketch thermal_cam that uses the I2C interface with these libraries:

#include <Adafruit_GFX.h>    // Core graphics library
#include <Adafruit_ST7735.h> // Hardware-specific library
#include <SPI.h>
#include <Wire.h>
#include <Adafruit_AMG88xx.h>

It compiles and runs fine.

The LIDARLite_v3HP library compiles and runs fine with the example sketch V3HP_PWM. I would like to use the LIDARLite_v3HP library for the I2C interface with the v3HP if possible since it gives more control than the PWM interface. I realize that the LIDARLite_v3HP library was designed to work with the Arduino Uno and that the MKR WIFI 1010 board may be incompatible with it, but I was hoping that this was not the case. Thanks for any help that you can provide.

[lidarcompileerrors.txt](https://github.com/garmin/LIDARLite_Arduino_Library/files/5659005/lidarcompileerrors.txt)

maulepilot, 2020-12-08:
I realized that I was still using the 3.0.5 LIDARLite_v3.cpp file that the Arduino IDE Library Manager downloads when you get the latest version. So I downloaded the 3.0.6 LIDARLite_v3.cpp file from github and renamed the old file. I also deleted files from the previous compile/build from /var/ since the compiler was trying to use them rather than re-compile. It still won't compile but now it seems to be trying to use the LIDARLite_v4LED.cpp file which probably wasn't changed. I don't know why it's trying to use that file since the v3HP_I2C sketch only has #include LIDARLite_v3HP.h and no mention of the v4. I attached the new compiler error output here.
[lidarcompileerrors.txt](https://github.com/garmin/LIDARLite_Arduino_Library/files/5660686/lidarcompileerrors.txt)

BradWiseman, 2020-12-08:
@maulepilot, I think you may have updated the library before the Arduino Library Manager had updated its access to all of the files that are involved in version 3.06.  Please try again and make sure that you get the LIDAR-Lite v4LED files too.

maulepilot, 2020-12-08:
It worked! Complled properly and ran. Now I need to figure out why I'm getting a bunch of nacks from the v3HP. I may have to change the value of the pull-up resistors from 4.7K. Thanks.

maulepilot, 2020-12-17:
BradWiseman, this compile problem returned after I updated my Arduino MKR WiFi 1010 board from 1.8.9 to 1.8.10. The updater deletes the existing ~/Library/Arduino15/packages/arduino/hardware/samd/1.8.9/libraries folder and replaces it with ~/Library/Arduino15/packages/arduino/hardware/samd/1.8.10/libraries. I re-created the platform.local.txt file in that directory with the single line compiler directive compiler.cpp.extra_flags=-DLEGACY_I2C. That didn't fix the problem. I deleted your LIDARLite_Arduino_Library-master directory and re-installed version 3.0.6 with the Arduino IDE library manager. That didn't fix the problem either. Any advice about what to try next? Attached is the verbose compiler output.
[lidarerrors.txt](https://github.com/garmin/LIDARLite_Arduino_Library/files/5712288/lidarerrors.txt)

maulepilot, 2020-12-19:
I modified the LIDARLite_v3HP.cpp and LIDARLite_v4LED.cpp library files to include the line #define LEGACY_I2C under the #include directives and the library compiles and executes properly now.

The method that you commented in these files of creating a platform.local.txt file with the compiler directive compiler.cpp.extra_flags=-DLEGACY_I2C doesn't work.

BradWiseman, 2020-12-22:
@maulepilot, glad to hear that you were able to modify Garmin's example code to fit your needs.  That's the purpose of their LIDARLite_Arduino_Library - to serve as examples of how to interface with Garmin's LIDAR-Lite series sensors.

I think that the more flexible way to fix the issue would be to create a text file named "platform.local.txt" that contains the following text:

compiler.cpp.extra_flags=-DLEGACY_I2C

That file should be stored to this folder on your machine:

/Users/ameranda/Library/Arduino15/packages/arduino/hardware/samd/1.8.10

Please note that different users will have a different path, depending on the folder that the Arduino IDE uses for compiling the project for your board.  For instance, I stored the platform.local.txt file to my C:\Users\wiseman\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.10 folder.  (I'm on a Windows machine.)  The way you can identify the correct folder into which the platform.local.txt file should be placed is to:

1) Select the "Compilation" checkbox in your File/Preferences window and click OK.
2) Use File/Open, File/Open Recent, File Sketchbook or File Examples to select the code that you wish to compile.
3) Use Tools/Board to load the board for which you intend to compile the Sketch
4) Verify/Compile your Sketch.
5) If errors occur, search your compiler log for "Using board".  The following line will identify the folder into which the platform.local.txt file should be placed. You should also see a "platform.txt" file already in that folder.  If no errors occur, then you will not need the platform.local.txt file.

It's the platform.txt file that identifies which compiler and build flags are available for override.  In the case of the samd boards, we used the compiler.cpp.extra_flags flag to override with the -D directive that defines LEGACY_I2C.  According to the comments in the platform.txt file, that flag can be overridden in the platform.local.txt file, so that's the name of the file that we use to override the compiler.cpp.extra_flags flag and to ensure that LEGACY_I2C is defined.

I hope this helps.

Again, the most important thing to remember is that the LIDARLite_Arduino_Library is intended to provide examples of how to interface with Garmin's LIDAR-Lite series of sensors.  It's not intended to be a production-ready source of code for all variants of Arduino boards.  It's kind of a one size fits most sort of thing.

Yankoboy, 2021-04-07:
@BradWiseman , @maulepilot , the solution presented also solved compilation issues with Arduino Nano 33 BLE Sense board, Garmins LIDARLite v3HP, I2C example, using IDE 1.8.13 on Win10. Similar error output as observed by maulepilot.
Thank you all!

abedin17, 2024-02-23:
@BradWiseman @maulepilot : I am getting this error for esp 32 and could not find a way to solve : :c:\Users\Admin\Documents\Arduino\libraries\LIDAR-Lite\src\LIDARLite_v3HP.cpp: In member function 'void LIDARLite_v3HP::read(uint8_t, uint8_t*, uint8_t, uint8_t)':
c:\Users\Admin\Documents\Arduino\libraries\LIDAR-Lite\src\LIDARLite_v3HP.cpp:455:5: error: no matching function for call to 'TwoWire::requestFrom(uint8_t&, uint8_t&, uint8_t&, int, bool)'
     );
     ^
In file included from c:\Users\Admin\Documents\Arduino\libraries\LIDAR-Lite\src\LIDARLite_v3HP.cpp:27:
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.12\libraries\Wire\src/Wire.h:119:12: note: candidate: 'size_t TwoWire::requestFrom(uint16_t, size_t, bool)'
     size_t requestFrom(uint16_t address, size_t size, bool sendStop);
            ^~~~~~~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.12\libraries\Wire\src/Wire.h:119:12: note:   candidate expects 3 arguments, 5 provided
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.12\libraries\Wire\src/Wire.h:120:13: note: candidate: 'uint8_t TwoWire::requestFrom(uint16_t, uint8_t, bool)'
     uint8_t requestFrom(uint16_t address, uint8_t size, bool sendStop);
             ^~~~~~~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.12\libraries\Wire\src/Wire.h:120:13: note:   candidate expects 3 arguments, 5 provided
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.12\libraries\Wire\src/Wire.h:121:13: note: candidate: 'uint8_t TwoWire::requestFrom(uint16_t, uint8_t, uint8_t)'
     uint8_t requestFrom(uint16_t address, uint8_t size, uint8_t sendStop);
             ^~~~~~~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.12\libraries\Wire\src/Wire.h:121:13: note:   candidate expects 3 arguments, 5 provided
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.12\libraries\Wire\src/Wire.h:122:12: note: candidate: 'size_t TwoWire::requestFrom(uint8_t, size_t, bool)'
     size_t requestFrom(uint8_t address, size_t len, bool stopBit);
            ^~~~~~~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.12\libraries\Wire\src/Wire.h:122:12: note:   candidate expects 3 arguments, 5 provided
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.12\libraries\Wire\src/Wire.h:123:13: note: candidate: 'uint8_t TwoWire::requestFrom(uint16_t, uint8_t)'
     uint8_t requestFrom(uint16_t address, uint8_t size);
             ^~~~~~~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.12\libraries\Wire\src/Wire.h:123:13: note:   candidate expects 2 arguments, 5 provided
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.12\libraries\Wire\src/Wire.h:124:13: note: candidate: 'uint8_t TwoWire::requestFrom(uint8_t, uint8_t, uint8_t)'
     uint8_t requestFrom(uint8_t address, uint8_t size, uint8_t sendStop);
             ^~~~~~~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.12\libraries\Wire\src/Wire.h:124:13: note:   candidate expects 3 arguments, 5 provided
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.12\libraries\Wire\src/Wire.h:125:13: note: candidate: 'uint8_t TwoWire::requestFrom(uint8_t, uint8_t)'
     uint8_t requestFrom(uint8_t address, uint8_t size);
             ^~~~~~~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.12\libraries\Wire\src/Wire.h:125:13: note:   candidate expects 2 arguments, 5 provided
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.12\libraries\Wire\src/Wire.h:126:13: note: candidate: 'uint8_t TwoWire::requestFrom(int, int, int)'
     uint8_t requestFrom(int address, int size, int sendStop);
             ^~~~~~~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.12\libraries\Wire\src/Wire.h:126:13: note:   candidate expects 3 arguments, 5 provided
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.12\libraries\Wire\src/Wire.h:127:13: note: candidate: 'uint8_t TwoWire::requestFrom(int, int)'
     uint8_t requestFrom(int address, int size);
             ^~~~~~~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.12\libraries\Wire\src/Wire.h:127:13: note:   candidate expects 2 arguments, 5 provided
c:\Users\Admin\Documents\Arduino\libraries\LIDAR-Lite\src\LIDARLite_v4LED.cpp: In member function 'void LIDARLite_v4LED::read(uint8_t, uint8_t*, uint8_t, uint8_t)':
c:\Users\Admin\Documents\Arduino\libraries\LIDAR-Lite\src\LIDARLite_v4LED.cpp:426:5: error: no matching function for call to 'TwoWire::requestFrom(uint8_t&, uint8_t&, uint8_t&, int, bool)'
     );
     ^
In file included from c:\Users\Admin\Documents\Arduino\libraries\LIDAR-Lite\src\LIDARLite_v4LED.cpp:27:
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.12\libraries\Wire\src/Wire.h:119:12: note: candidate: 'size_t TwoWire::requestFrom(uint16_t, size_t, bool)'
     size_t requestFrom(uint16_t address, size_t size, bool sendStop);
            ^~~~~~~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.12\libraries\Wire\src/Wire.h:119:12: note:   candidate expects 3 arguments, 5 provided
[... cut ...]
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.12\libraries\Wire\src/Wire.h:126:13: note:   candidate expects 3 arguments, 5 provided
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.12\libraries\Wire\src/Wire.h:127:13: note: candidate: 'uint8_t TwoWire::requestFrom(int, int)'
     uint8_t requestFrom(int address, int size);
             ^~~~~~~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.12\libraries\Wire\src/Wire.h:127:13: note:   candidate expects 2 arguments, 5 provided

exit status 1

Compilation error: exit status 1